MJIrobotics/Tapia-sdk #2: 二点ご質問があります。

SakaguchiYoshiki, 2017-09-13:
お疲れ様です。
二点ご質問があります。

TOSチュートリアルに従い、タブレットでの動作確認まで行っい、タブレットにTapiaの顔が表示されました。
ここでタブレットのTapiaのエミュレータを終了（タスクキル）したところ、ホーム画面の壁紙がTapiaの顔になりました。
再起動してもそのままです。
これは仕様ですか？

タブレットでエミュレータが動いたので、
TOSのページにある
2.TAPIAに言葉を話させる
3.TAPIAに言葉を認識させる
などをやりたいのですが、
TapiaApp.setCurrentLanguage(Language.LanguageID.JAPANESE);
などのコードをどこに記述し、どう実行するのかが分かりません。

よろしくお願いします。

MJIrobotics, 2017-09-14:
壁紙がTapiaの顔なるという現象ですが
サンプルプロジェクトのStartActivity.javaの35行~36行に壁紙を変更する記述があります。
こちらをコメントアウトする等で対応してください。

また、2.TAPIAに言葉を話させるや3.TAPIAに言葉を認識させるコードを記述する箇所は
既にあるActivity(上記のStartActivity.java)、または新たにActiveityを作成して
実行する事も可能です。
その辺はご説明は一度AndroidStudioによる開発ガイダンスをご覧いただければと思います。

SakaguchiYoshiki, 2017-09-20:
ご回答いただきありがとうございます。

closed 2017-09-27 by MJIrobotics