MJIrobotics/Tapia-sdk #3: 画面の端を触ると黒くなる

IWAMIGINZI, 2017-09-20:
お世話になります。

アプリ起動中、画面の端の方に触れると
画面が狭くなります。(画面端に黒い枠のようなものが現れる?)
アプリをkillするまでずっと現れているのですが、
こちらの原因は何でしょうか。

MJIrobotics, 2017-09-26:
androidの機能によるもので、現在のＳＤＫでは対応ができない現象です。
次のバージョンでの改善を予定しております。
ご不便をおかけして申し訳ございませんが、しばらくお待ちくださいませ。
よろしくお願い申し上げます。

TakakoKubo, 2018-01-24:
SDKのバージョンはVer1.07となり、画面に表示される黒枠についても対応いたしました事をご連絡いたします。現在ダウンロードして使用できますので何卒よろしくお願い申し上げます。

closed 2018-01-24 by TakakoKubo

IWAMIGINZI, 2018-01-25:
対応ありがとうございます。
バージョン1.0.7のアップをしたいのですが、弊社のセキュリティ要件で接続できません。
前回同様、ローカル環境でver1.0.7を使えるように対応をお願い致します。

TakakoKubo, 2018-01-26:
■ダウンロードお願いします。
https://drive.google.com/open?id=17xYi5nyoitg6BfY9bDyx8i_VsDWYb2fQ
■設定手順は以下の通りです。AndroidStudioのプロジェクトを選択して進めてください。
ダウンロードが終わりましたら連絡いただけないでしょうか。どうぞよろしくお願いいたします。
1.Right click on your project and select "Open Module Settings".
2.Click the "+" button in the top left corner of window to add a new module.
3.Select "Import .JAR or .AAR Package" and click the "Next" button.
4.Find the AAR file using the ellipsis button "..." beside the "File name" field.
5.Keep the app's module selected and click on the Dependencies pane to add the new module as a dependency.
6.Use the "+" button of the dependencies screen and select "Module dependency".
7.Select the module and click "OK".

IWAMIGINZI, 2018-01-26:
ご対応ありがとうございます。
6の手順の、"Module dependency"で選択肢が出てこないのですが、
これは何が原因でしょうか。
ファイルの置き場所が悪いのでしょうか。
以上、よろしくお願い致します。

TakakoKubo, 2018-01-29:
手順の確認をさせていただきたいのですがよろしいでしょうか。
画像付マニュアルを添付しますので操作について確認していただけないでしょうか。
お手数おかけしますがよろしくお願いいたします。

[マニュアル.docx](https://github.com/MJIrobotics/Tapia-sdk/files/1671826/default.docx)

IWAMIGINZI, 2018-01-29:
添付いただいたマニュアル通りに再度操作致しました。
しかし、マニュアルの7手順目の「Choose Modules」の画面で選択肢が出てきません。
以上、よろしくお願い致します。

TakakoKubo, 2018-01-29:
以下の手順で追加をお願いいたします。
①プロジェクトのソースが置いてあるapp下に、「libs」フォルダを作成します。
\tapia-sample\app\libs

②その「libs」下にダウンロードした「tapialib-1-1.0.7.aar」をコピーします。
\tapia-sample\app\libs\tapialib-1-1.0.7.aar

③「build.grade」のdependenciesに以下の修正をします。
dependencies {

//    compile 'com.tapia.mji:tapialib:1.0.XX'　　←コメントアウトします。
    compile(name:'tapialib-1-1.0.7', ext:'aar'　←追加します。
)
}

IWAMIGINZI, 2018-01-29:
DLしたファイル名が違ったので「tapialib-1.0.7.aar」に直して手順通り進めました。
エラーは出ず、TAPIAも正常に動きます。
しかし、画面の端を触るとまだ黒い枠のようなものが出ています。
うまくいっていないのでしょうか。
以上、よろしくお願い致します。

TakakoKubo, 2018-01-30:
現在、現象を検証しております。確認後ご連絡いたしますので少しお時間いただけないでしょうか。
どうぞよろしくお願いいたします。

TakakoKubo, 2018-02-01:
以下のよりダウンロードして設定の上使用していただけないでしょうか。
ダウンロード
https://tos.mjirobotics.co.jp/sdk/download/tapia_sdk_v_1.0.7.zip
マニュアルの設定方法(2)～を参照してください。
https://tos.mjirobotics.co.jp/sdk/manual/4.html

IWAMIGINZI, 2018-02-01:
下記のエラーが出ます。

Error:(35, 13) Failed to resolve: com.tapia.mji:tapialib:1.0.7
Show in File
Show in Project Structure dialog

以上、よろしくお願い致します。

TakakoKubo, 2018-02-01:
エラーが出るのはどのタイミングか教えていただけないでしょうか。
どうぞよろしくお願いいたします。

IWAMIGINZI, 2018-02-01:
最初からです。
手順の「2)ライブラリを開く」です。

TakakoKubo, 2018-02-02:
下記確認させてください
確認①最初に案内したbuild.gradeの設定を元にもどしましたでしょうか。
「build.grade」のdependenciesに以下の修正をします。
dependencies {
compile 'com.tapia.mji:tapialib:1.0.XX'　　←元に戻します。
//compile(name:'tapialib-1-1.0.7', ext:'aar' ←コメントアウトします。
確認②先ほどコピーしたlibsフォルダ下のtapialib-1-1.0.7.aarを削除していただけましたでしょうか。
\tapia-sample\app\libs\tapialib-1-1.0.7.aar　←削除
確認②ブラックバーについて
画面端に表示する黒い枠のようなものというのがこちらの認識と相違があるのか確認させてください。
もしも画像をとる事ができたら送っていただくことできますか。
その黒枠は画面下のみに出る黒枠ですか?

IWAMIGINZI, 2018-02-02:
確認①②
https://tos.mjirobotics.co.jp/sdk/download/tapia_sdk_v_1.0.7.zip
をダウンロードし、
https://tos.mjirobotics.co.jp/sdk/manual/4.html
マニュアルに沿って、tapia-sampleをAndroidStudioでそのまま開いているので、
そもそもbuild.gradeの設定などはしていませんし、
libsフォルダも作成しておりません。
確認③
以下です。
[image]

TakakoKubo, 2018-02-06:
この度は画像をお送りいただきありがとうございます。
大変申し訳ありません、お送りいただいたブラックバーの不具合はソースコードを追加する必要があります。ご案内が遅れ申し訳ありませんでした。
以下の最初にご案内した手順でエラーがでないとの事でしたので、
その設定の後、ソースコードを追加してブラックバーの表示、非表示について問題がないか確認していただけないてしようか。
■最初の案内
　　①プロジェクトのソースが置いてあるapp下に、「libs」フォルダを作成します。
　　\tapia-sample\app\libs
　　②その「libs」下にダウンロードした「tapialib-1-1.0.7.aar」をコピーします。
　　\tapia-sample\app\libs\tapialib-1-1.0.7.aar　　
　　③「build.grade」のdependenciesに以下の修正をします。
　　dependencies {　
　　// compile 'com.tapia.mji:tapialib:1.0.XX'　　←コメントアウトします。
　　compile(name:'tapialib-1-1.0.7', ext:'aar'　←追加します。
　　)}
■追加するソースコード
StartActivityの「onCreate」メソッドに以下を追加します。
>>TapiaApp.hideAndroidUi(false); //trueブラックバー表示、false非表示

なお、Error:(35, 13)…のエラーは、最初の設定が影響している場合に発生します。
上記の設定でうまくいかない場合、再度ご相談させてください。

何か不明点ありましたら連絡くださいますようお願い申し上げます。
色々とお手数おかけして大変申し訳ありませんがどうぞよろしくお願いいたします。

IWAMIGINZI, 2018-02-06:
ソースコードを追加し、ブラックバーが消えることを確認致しました。
ご回答ありがとうございました。