MJIrobotics/Tapia-sdk #6: 音に反応し、音のする方向へ回転します。

HiroakiKanda, 2017-09-28:
いつもお世話になります。

製品版タピアは
待機モードのとき、音に反応し、音のする方向へ回転します。
これを開発SDKで実現する方法を教えてください。

岩谷情報システム神田

MJIrobotics, 2017-10-06:
回答が遅れてすみませんでした。
SDKのライブラリーを確認したところ

//音の聞こえる方向にタピアを合わせるモードの開始
static void startSoundLocation(android.content.Context context)

//音の聞こえる方向にタピアを合わせるモードの停止
static void stopSoundLocation(android.content.Context context)

この機能は自動的に音の聞こえる方向に回転するだけで
その方向をとらえる事は出来ません。

以上、よろしくお願いします。

closed 2017-10-24 by MJIrobotics

HiroakiKanda, 2018-03-30:
以前に教えていただきましたstartSoundLocation、stopSoundLocationは
どのクラスのメソッドなのか教えてください。

//音の聞こえる方向にタピアを合わせるモードの開始
static void startSoundLocation(android.content.Context context)

//音の聞こえる方向にタピアを合わせるモードの停止
static void stopSoundLocation(android.content.Context context)

岩谷情報システム 神田