MJIrobotics/Tapia-sdk #8: TAPIA Animationの自動切換え

IWAMIGINZI, 2017-10-06:
お世話になっております。

一定時間経過後、TAPIAのAnimationを違うAnimationに切り替えたいと考えております。
(例. ALARMAnimation → x分後 → CRYINGAnimation →x分後 →  SMILINGAnimation → ……)

現在、Timerクラスで上記の動きを実装しているのですが、
別Activityから、Animationを行うActivityに遷移することを繰り返すと、
Animationの動きが段々と鈍くなってしまいます。

Animationを行うActivity内の
onResume()でTimerクラスを実装しており、
onPause()で　tapiaAnimation.stopAnimation();
を記述しております。

こちら、どうにかAnimationが鈍くならない方法はないでしょうか。

そもそもTimerクラスを使わなくても、TapiaAnimationのメソッドで
実装できるようであれば、その方法もお教えいただきたいです。

以上、よろしくお願い致します。

IWAMIGINZI, 2017-10-06:
お世話になっております。

アニメーションが鈍くなる件、解決致しました。
timerを停止させていなかったことが問題でした。

Timerクラスを使わなくても、TapiaAnimationで実装できるかどうかに
関しましては、引き続きご回答お待ちしております。

以上、よろしくお願い致します。

MJIrobotics, 2017-10-11:
お世話になります。

TapiaAnimationが遅くなる状態は、恐らくご指摘の通り新しいAnimationを開始する際に
旧AnimationのTimerを停止していなかった為に、旧Animationも他の新しいAnimationを読んでいた可能性があり、結果的にAnimationが複数起動している状態になっていたのではないかと推察されます。

現在のTapiaAnimationは繰り返し処理の際にはTimerと併用することでしか実装できません。
TapiaAnimationを拡張子ご要望の機能を追加する案もありましたが、現在たの改修を優先しているため
ご要望の機能の実装は具体的に予定していません。

よろしくお願いします。

closed 2017-10-11 by MJIrobotics