MJIrobotics/Tapia-sdk #10: 予期せぬエラーでアプリケーションが停止します

IWAMIGINZI, 2017-10-13:
お世話になっております。

製品版及び開発版のTAPIAでアプリケーションを起動し続けていると
アプリケーションが終了してしまいます。
およそ1時間ほどで終了してしまうのですが、
こちらの原因はなんでしょうか。
また、原因を特定する方法があれば教えていただきたいです。

以上、よろしくお願い致します。

IWAMIGINZI, 2017-10-13:
アプリケーション起動中にAndroidMonitorに以下のログが常時出力されます。
以下アプリケーションが予期せぬエラーとなるのに関係あるでしょうか？
・W/System.err: java.io.FileNotFoundException: f6_/f6_00100.png
　→f6_/f6_00100.pngのファイルは使用していない。
・/com.tapia.mji.demo D/Surface: Surface::setBuffersDimensions(this=0xf49cec00,w=720,h=1280)
・W/MALI: glDrawArrays:714: [MALI] glDrawArrays takes more than 5ms here. Total elapse time(us): 5850
　→これはなにをしているのか？

10-13 09:29:54.357 9163-9163/com.tapia.mji.demo W/System.err: java.io.FileNotFoundException: f6_/f6_00100.png
10-13 09:29:54.358 9163-9163/com.tapia.mji.demo W/System.err:     at android.content.res.AssetManager.openAsset(Native Method)
10-13 09:29:54.358 9163-9163/com.tapia.mji.demo W/System.err:     at android.content.res.AssetManager.open(AssetManager.java:318)
10-13 09:29:54.358 9163-9163/com.tapia.mji.demo W/System.err:     at android.content.res.AssetManager.open(AssetManager.java:292)
10-13 09:29:54.358 9163-9163/com.tapia.mji.demo W/System.err:     at com.tapia.mji.tapialib.Utils.TapiaAnimation.getBitmap(TapiaAnimation.java:563)
10-13 09:29:54.358 9163-9163/com.tapia.mji.demo W/System.err:     at com.tapia.mji.tapialib.Utils.TapiaAnimation.access$600(TapiaAnimation.java:32)
10-13 09:29:54.358 9163-9163/com.tapia.mji.demo W/System.err:     at com.tapia.mji.tapialib.Utils.TapiaAnimation$1$1.run(TapiaAnimation.java:374)
10-13 09:29:54.358 9163-9163/com.tapia.mji.demo W/System.err:     at android.os.Handler.handleCallback(Handler.java:815)
10-13 09:29:54.358 9163-9163/com.tapia.mji.demo W/System.err:     at android.os.Handler.dispatchMessage(Handler.java:104)
10-13 09:29:54.358 9163-9163/com.tapia.mji.demo W/System.err:     at android.os.Looper.loop(Looper.java:194)
10-13 09:29:54.358 9163-9163/com.tapia.mji.demo W/System.err:     at android.app.ActivityThread.main(ActivityThread.java:5643)
10-13 09:29:54.358 9163-9163/com.tapia.mji.demo W/System.err:     at java.lang.reflect.Method.invoke(Native Method)
10-13 09:29:54.358 9163-9163/com.tapia.mji.demo W/System.err:     at java.lang.reflect.Method.invoke(Method.java:372)
10-13 09:29:54.358 9163-9163/com.tapia.mji.demo W/System.err:     at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:960)
10-13 09:29:54.358 9163-9163/com.tapia.mji.demo W/System.err:     at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:755)[](url)

MJIrobotics, 2017-10-13:
IWAMIGINZI 様

　
ご指摘の　「W/System.err: java.io.FileNotFoundException: f6_/f6_00100.png」から
始まる箇所はタピアアニメーション時に0から始まるファイルを順次表示し
100個目のファイル(f6_00100.png)でファイルが存在しないExceptionを取得して
最初に戻る際に吐き出されるログです。
本来はExceptionログとせず正常終了すべきなので今後改善をいたします。

また、起動後1時間程度でタピアが終了してしまうとのご指摘ですが、その際のエラーログや
動画をキャプションしていただき、アップロードしていただくと助かります。

よろしくお願いします。

closed 2017-10-24 by MJIrobotics