MJIrobotics/Tapia-sdk #11: TAPIAと会話をしたい

HiroakiKanda, 2017-10-13:
お世話になります。

製品版のようにTAPIAと会話をしたいです。

サンプルコードの
3. TAPIAに、言葉を言葉として認識させる
4. TAPIAに言葉の内容を理解させる
を見ましたが理解が難しいです。

全体の流れがわかるような資料や
サンプルソースなど提供いただけないでしょうか。

岩谷情報システム 神田

MJIrobotics, 2017-10-13:
ご連絡ありがとうございます。

詳細サンプルコードを提示するよう準備をいたします。
しばらくお待ちください。

HiroakiKanda, 2017-10-31:
>詳細サンプルコードを提示するよう準備をいたします。
状況いかがでしょうか？
いつごろサンプルコードを提示いただけるか教えてください。

岩谷情報システム 神田

HiroakiKanda, 2017-11-15:
> From    ： MJIrobotics <notifications@github.com>
> Date    ： 2017/10/13 18:30:38

MJI 井口様
いつもお世話になります。
岩谷情報システム 神田です。

回答いただいてから１ヶ月経過しましたが状況いかがでしょうか？

だいたいの提供時期だけでも教えてください。

以上、よろしくお願い致します。

TakakoKubo, 2017-11-29:
岩谷情報システム神田様

MJI窪と申します。
新しくSDKを担当しております。
さて、この度は回答遅くなりまして大変申し訳ありませんでした。
問い合わせのソースについては現在作成対応しておりまして
今週金曜日には提供時期のめどをつけまして再度ご連絡したいと思います。
お時間をおかけしました事大変申し訳ありませんでした。
今後ともどうぞよろしくお願い申し上げます。

HiroakiKanda, 2017-11-30:
> From    ： TakakoKubo <notifications@github.com>
> Date    ： 2017/11/29 10:07:14

MJI 窪様
いつもお世話になります。
岩谷情報システム 神田です。

回答ありがとうございます。

どうぞよろしくお願いいたします。

以上です。

TakakoKubo, 2017-12-01:
岩谷情報システム神田様
いつもお世話になっております。
遅くなり大変申し訳ありませんでした。
サンプルソースですが、本日中に提供したく現在最終テストをしております。
恐れ入りますがあと少しお待ちいただきたくどうぞよろしくお願いいたします。

HiroakiKanda, 2017-12-01:
承知致しました。
よろしくお願いいたします。

神田

TakakoKubo, 2017-12-01:
岩谷情報システム神田様
いつもお世話になっております。
この度はお待たせして大変申し訳ありませんでした。
以下の通りサンプルソースをご連絡いたします。
何か不明点ありましたら連絡くださいますようお願い申し上げます。
--------------------------------------------------
3. TAPIAに、言葉を言葉として認識させる方法
--------------------------------------------------
※例: タピアに話かけると、その同じ言葉をタピアが返す
※サンプルコード
「実装したいActivity.CLASS」の「onCreate」メソッド内へ以下を追加します。
※このActivity.CLASSが起動した時に話しかけます。
        TapiaApp.setCurrentLanguage(Language.LanguageID.JAPANESE);
        sttProvider = TapiaApp.currentLanguage.getOnlineSTTProvider();
        ttsProvider = TapiaApp.currentLanguage.getTTSProvider();
        sttProvider.listen();

        sttProvider.setOnRecognitionCompleteListener(new STTProvider.OnRecognitionCompleteListener() {

            public void onRecognitionComplete(List<String> list) {
                    try {
                        ttsProvider.say(list.get(0).toString());
                    } catch (LanguageNotSupportedException e) {
                        e.printStackTrace();
                    }
                    ttsProvider.setOnSpeechCompleteListener(new TTSProvider.OnSpeechCompleteListener() {

                        public void onSpeechComplete() {
                            sttProvider.stopListening();
                        }
                    });
                }

        });

--------------------------------------------------
4. TAPIAに言葉の内容を理解させる
--------------------------------------------------
※例:タピアに事前に登録した言葉を話しかけると、事前に登録した言葉をタピアが返す
  例えば:タピアに「名前」と聞くと、「タピアです」と答える

①MyAction.javaクラスの「enum MyActionType」へ「INTRODUCE」を追加します。
※サンプルコード
   public enum MyActionType implements ActionType {
        CONVERSATION,
        ROTATE,
        GIVE_TIME,
        GIVE_DATE,
        BYE,
        INTRODUCE,　<<追加します
    }

②Local_NLU.javaクラスの「Local_NLU」へ「名前」(Tapiaに認識して欲しい言葉)を追加します。
※サンプルコード
    public Local_NLU(Context context, Language.LanguageID language){
        this.language = language;
        this.context = context;
        if(language == Language.LanguageID.ENGLISH_US || language == Language.LanguageID.ENGLISH_UK) {
            myKeywords.add(new Keyword(new String[]{"what", "time"}, GIVE_TIME));
            myKeywords.add(new Keyword(new String[]{"rotate", "degree"}, MyAction.MyActionType.ROTATE));
        }
        else if(language == Language.LanguageID.JAPANESE){
            myKeywords.add(new Keyword(new String[]{"何時"}, MyAction.MyActionType.GIVE_TIME));
            myKeywords.add(new Keyword(new String[]{"何曜日"}, MyAction.MyActionType.GIVE_DATE));
            myKeywords.add(new Keyword(new String[]{"何日"}, MyAction.MyActionType.GIVE_DATE));
            myKeywords.add(new Keyword(new String[]{"回転", "度"}, MyAction.MyActionType.ROTATE));
            myKeywords.add(new Keyword(new String[]{"名前"}, MyAction.MyActionType.INTRODUCE));<<追加します

        }
    }

③MySimpleAction.javaクラスに、以下のクラスを追加します。
※サンプルコード
    public static class Introduce extends MySimpleAction {
        public Introduce(SimpleAction.OnSimpleActionListener simpleActionListener) {
            super(simpleActionListener, MyActionType.INTRODUCE);
        }
    }

④「実装したいActivity.CLASS」の「onCreate」メソッド内へ以下を追加します
※このActivity.CLASSが起動した時に話しかけます。
※サンプルコード
      TapiaApp.setCurrentLanguage(Language.LanguageID.JAPANESE);
        sttProvider = TapiaApp.currentLanguage.getOnlineSTTProvider();
        ttsProvider = TapiaApp.currentLanguage.getTTSProvider();
        offlineNLUProvider = TapiaApp.currentLanguage.getOfflineNLUProvider();
        final ArrayList actions = new ArrayList<>();

        actions.add(new MySimpleAction.Introduce(new SimpleAction.OnSimpleActionListener() {
            @Override
            public void onSimpleAction() {
                try {
                    ttsProvider.ask("私の名前はタピアです", sttProvider);
                } catch (LanguageNotSupportedException e) {
                    e.printStackTrace();
                }
                     ttsProvider.setOnSpeechCompleteListener(null);
            }
            }));

        sttProvider.listen();
        sttProvider.setOnRecognitionCompleteListener(new STTProvider.OnRecognitionCompleteListener() {
            @Override
            public void onRecognitionComplete(List<String> list) {
                offlineNLUProvider.analyseText(list, actions);
                ttsProvider.setOnSpeechCompleteListener(new TTSProvider.OnSpeechCompleteListener() {
                    @Override
                    public void onSpeechComplete() {
                        sttProvider.stopListening();
                    }
                });
            }
        });

TakakoKubo, 2017-12-01:
岩谷情報システム神田様
いつもお世話になっております。
この度はお待たせして大変申し訳ありませんでした。
以下の通りサンプルソースをご連絡いたします。
何か不明点ありましたら連絡くださいますようお願い申し上げます
…………………………………………………………………
3. TAPIAに、言葉を言葉として認識させる方法
…………………………………………………………………
※例: タピアに話かけると、その同じ言葉をタピアが返す
※サンプルコード
「実装したいActivity.CLASS」の「onCreate」メソッド内へ以下を追加します。

        TapiaApp.setCurrentLanguage(Language.LanguageID.JAPANESE);
        sttProvider = TapiaApp.currentLanguage.getOnlineSTTProvider();
        ttsProvider = TapiaApp.currentLanguage.getTTSProvider();
        sttProvider.listen();

        sttProvider.setOnRecognitionCompleteListener(new STTProvider.OnRecognitionCompleteListener() {

            public void onRecognitionComplete(List<String> list) {
                    try {
                        ttsProvider.say(list.get(0).toString());
                    } catch (LanguageNotSupportedException e) {
                        e.printStackTrace();
                    }
                    ttsProvider.setOnSpeechCompleteListener(new TTSProvider.OnSpeechCompleteListener() {

                        public void onSpeechComplete() {
                            sttProvider.stopListening();
                        }
                    });
                }

        });

…………………………………………………………………
4. TAPIAに言葉の内容を理解させる
…………………………………………………………………
※例:タピアに事前に登録した言葉を話しかけると、事前に登録した言葉をタピアが返す
  例えば:タピアに「名前」と聞くと、「タピアです」と答える

①MyAction.javaクラスの「enum MyActionType」へ「INTRODUCE」を追加します。
※サンプルコード
   public enum MyActionType implements ActionType {
        CONVERSATION,
        ROTATE,
        GIVE_TIME,
        GIVE_DATE,
        BYE,
        INTRODUCE,　<<追加します
    }

②Local_NLU.javaクラスの「Local_NLU」へ「名前」(Tapiaに認識して欲しい言葉)を追加します。
※サンプルコード
    public Local_NLU(Context context, Language.LanguageID language){
        this.language = language;
        this.context = context;
        if(language == Language.LanguageID.ENGLISH_US || language == Language.LanguageID.ENGLISH_UK) {
            myKeywords.add(new Keyword(new String[]{"what", "time"}, GIVE_TIME));
            myKeywords.add(new Keyword(new String[]{"rotate", "degree"}, MyAction.MyActionType.ROTATE));
        }
        else if(language == Language.LanguageID.JAPANESE){
            myKeywords.add(new Keyword(new String[]{"何時"}, MyAction.MyActionType.GIVE_TIME));
            myKeywords.add(new Keyword(new String[]{"何曜日"}, MyAction.MyActionType.GIVE_DATE));
            myKeywords.add(new Keyword(new String[]{"何日"}, MyAction.MyActionType.GIVE_DATE));
            myKeywords.add(new Keyword(new String[]{"回転", "度"}, MyAction.MyActionType.ROTATE));
            myKeywords.add(new Keyword(new String[]{"名前"}, MyAction.MyActionType.INTRODUCE));<<追加します

        }
    }

③MySimpleAction.javaクラスに、以下のクラスを追加します。
※サンプルコード
    public static class Introduce extends MySimpleAction {
        public Introduce(SimpleAction.OnSimpleActionListener simpleActionListener) {
            super(simpleActionListener, MyActionType.INTRODUCE);
        }
    }

④「実装したいActivity.CLASS」の「onCreate」メソッド内へ以下を追加します
※サンプルコード
      TapiaApp.setCurrentLanguage(Language.LanguageID.JAPANESE);
        sttProvider = TapiaApp.currentLanguage.getOnlineSTTProvider();
        ttsProvider = TapiaApp.currentLanguage.getTTSProvider();
        offlineNLUProvider = TapiaApp.currentLanguage.getOfflineNLUProvider();
        final ArrayList actions = new ArrayList<>();

        actions.add(new MySimpleAction.Introduce(new SimpleAction.OnSimpleActionListener() {
            @Override
            public void onSimpleAction() {
                try {
                    ttsProvider.ask("私の名前はタピアです", sttProvider);
                } catch (LanguageNotSupportedException e) {
                    e.printStackTrace();
                }
                     ttsProvider.setOnSpeechCompleteListener(null);
            }
            }));

        sttProvider.listen();
        sttProvider.setOnRecognitionCompleteListener(new STTProvider.OnRecognitionCompleteListener() {
            @Override
            public void onRecognitionComplete(List<String> list) {
                offlineNLUProvider.analyseText(list, actions);
                ttsProvider.setOnSpeechCompleteListener(new TTSProvider.OnSpeechCompleteListener() {
                    @Override
                    public void onSpeechComplete() {
                        sttProvider.stopListening();
                    }
                });
            }
        });

TakakoKubo, 2017-12-01:
先ほど追加したコメントですが、文字フォントが大きく掲載されてしまい、なかなか元に戻らなかった為、何度か追記・記載いたしました。大変申し訳ありませんでした。

HiroakiKanda, 2017-12-01:
いつもお世話になります。
岩谷情報システム 神田です。

サンプルコードの提供ありがとうございます。

不明点あれば問合せ致します。

以上、よろしくお願い致します。

Date    ： 2017/12/01 15:44:29

noorHariz, 2017-12-19:
Can I know, what is the method that should I do in order to make the Tapia speak English? For now, I already have the mji account. I hope you can help me. Thanks

TakakoKubo, 2017-12-25:
大変申し訳ありません。
現在SDKではEnglish版に対応してありません。
今後については、追加する方向で検討したいと思います。
恐れ入りますがどうぞよろしくお願いいたします。

HiroakiKanda, 2018-01-15:
お世話になります。
サンプルコードの問合せです。
------------------------
確認できたこと
------------------------
１）3. TAPIAに、言葉を言葉として認識させる方法
⇒sttProvider.stopListening();をsttProvider.listen();に変更するこおとで
　話した言葉を繰り返し話すようになりました。期待の動きをしました。

２）4. TAPIAに言葉の内容を理解させる
⇒Local_NLUへ登録した内容は応答することが確認できました。

------------------------
質問・実装したいこと
------------------------
１）4. TAPIAに言葉の内容を理解させる
⇒Local_NLUに登録がないとlisten()が終了してしまう。
　登録がない場合は、「わかりません」と応答してlisten()を継続させたいのですが
　どのように実装すれば良いでしょうか？

　または、登録がない場合は3の話した言葉を繰り返し話すように
　することは可能でしょうか？

２）サンプルではofflineのNLUを使用していますが
クラウドなどonlineのNLUを利用することは可能でしょうか？

岩谷情報システム 神田

HiroakiKanda, 2018-01-18:
お世話になります。

３日前に問合せしましたが
確認いただけましたでしょうか？

岩谷情報システム 神田

TakakoKubo, 2018-01-18:
ご返信遅くなり申し訳ありません。
現在確認中です。時間をいただき大変恐れ入りますが、確認でき次第ご連絡いたしますので何卒よろしくお願いいたします。

TakakoKubo, 2018-01-22:
遅くなり大変恐れ入ります。以下についてご確認をお願いいたします。よろしくお願いいたします。
1)について、
「4. TAPIAに言葉の内容を理解させる」の
④「実装したいActivity.CLASS」の「onCreate」メソッド内の
setOnRecognitionCompleteListenerに追加します。
以下サンプル例ですので、要件にあわせて追加修正の上使用して下さい。　

           @Override
            public void onRecognitionComplete(List<String> list) {
                //↓↓↓追加します↓↓↓
                offlineNLUProvider.setOnAnalyseCompleteListener(new NLUProvider.OnAnalyseCompleteListener() {
                    @Override
                    public void OnAnalyseComplete(Action action) {
                        if (action == null) {
                            try {
                                ttsProvider.say("わかりません");
                            } catch (LanguageNotSupportedException e) {
                                e.printStackTrace();
                            }
                        }

                    }
                });
                //↑↑↑追加します↑↑↑
                offlineNLUProvider.analyseText(list, actions);
                ttsProvider.setOnSpeechCompleteListener(new TTSProvider.OnSpeechCompleteListener() {
                    @Override
                    public void onSpeechComplete() {
                        sttProvider.stopListening();
                    }
                });
            }

2)について、
クラウドなどonlineのNLUを利用すること可能です。
市販で色々出ていると思いますので、そちらも利用することができると思います。

HiroakiKanda, 2018-01-22:
回答ありがとうございます。

1)については
こちらで実装してみます。

2)について、
>クラウドなどonlineのNLUを利用すること可能です。
>市販で色々出ていると思いますので、そちらも利用することができると思います。

御社が実際に利用しているサービスなど
具体的な情報があればご教授いただけないでしょうか
どうぞよろしくお願いいたします。

岩谷情報システム 神田

TakakoKubo, 2018-01-23:
こちらの説明が不足しており大変申し訳ありませんでした。
市販のサービスを利用することは可能と思いますが、
弊社では、市販のサービスは利用しておりませんので、
情報をもっておらず具体的な情報の提供は厳しい状態です。

また現在、SDKでは標準はNLUはofflineとなっておりますが、
Androidアプリなので、要件に合わせてカスタマイズして利用することが可能です。

それではどうぞよろしくお願いいたします。

HiroakiKanda, 2018-01-23:
製品版タピアの会話機能は
offlineのNLUで稼働しているということでしょうか
または独自の会話システムがあるのでしょうか

会話を学習して成長していくタピアを
実現したいと考えています。

岩谷情報システム 神田

TakakoKubo, 2018-01-23:
製品版タピアですが、
onlineとofflineのNLU両方で稼働しております。onlineは独自にDBを作成して運用しております。
それではどうぞよろしくお願いいたします。

HiroakiKanda, 2018-01-30:
製品版タピアのonline版を弊社も有償で利用可能でしょうか？

神田

HiroakiKanda, 2018-02-02:
お世話になります。
3日前に問合せしましたが
受付していただいたのか、放置されているのかがわかりません。
回答に時間を有する場合でも
一旦は受付ましたと回答いただけると
ユーザーは安心感があります。

岩谷情報システム　神田

TakakoKubo, 2018-02-02:
この度は遅くなり大変申し訳ありませんでした。
現在、製品版タピアのonline版の有償利用の提供は行っておりませんが、今後提供するかについては検討中です。時期については未定でありますが、状況に変化があり報告できる場合ありましたらご連絡したいと思います。恐れ入りますがどうぞよろしくお願い申し上げます。

closed 2018-04-26 by MJIrobotics