MJIrobotics/Tapia-sdk #15: 音声認識結果（テキストファイル）と画像ファイルの格納先を教えてください

satotakehirohack, 2017-11-25:
TAPIAが音声認識した結果のテキストファイルと、撮影した画像をファイルを取得したいのですがどこに保存されていますか。またどのようなコードを書けばよいでしょうか。

TakakoKubo, 2017-11-29:
音声認識した結果テキストファイルですが、現在、保存する設定となっておりません。
また撮影した画像ファイルについては、確認出来次第ご連絡いたしますので
お時間をいたたき大変恐れ入りますが、何卒引き続きよろしくお願いいたします。

satotakehirohack, 2017-11-29:
ご回答ありがとうございます。
テキスト変換前の音声ファイルも保存されないでしょうか。
お忙しいところ恐縮ですがご回答頂けると幸いです。

TakakoKubo, 2017-12-01:
画像ファイルについて、
CameraHelperクラスに、下記の通り画像ファイルの保存場所を設定しています。
・public static final String SMALL_PICTURE_FOLDER = Environment.getExternalStorageDirectory().toString() + "/RobotPhoto_Small/";
・public static final String PICTURE_FOLDER = Environment.getExternalStorageDirectory().toString() + "/RobotPhoto/";
タピアをUSBで接続して、エクスプローラーからそのフォルダを選択しても画像の確認をすることができます。

テキスト変換前の音声ファイルは、保存されないのですが
STTProvider　interfaceの、以下メソッドを使う事により取得することができます。
・void setOnRecognitionCompleteListener(STTProvider.OnRecognitionCompleteListener var1);

・記述例
TapiaApp.setCurrentLanguage(Language.LanguageID.JAPANESE);
sttProvider = TapiaApp.currentLanguage.getOnlineSTTProvider();
ttsProvider = TapiaApp.currentLanguage.getTTSProvider();
sttProvider.listen();
sttProvider.setOnRecognitionCompleteListener(new STTProvider.OnRecognitionCompleteListener() {
    @Override
    public void onRecognitionComplete(List<String> list) {
        for (String s: list) {
            try {
                ttsProvider.say(s);
            } catch (LanguageNotSupportedException e) {
                e.printStackTrace();
            }
それでは何卒よろしくお願い申し上げます。

satotakehirohack, 2017-12-02:
ありがとうございます。トライしてみます。

TakakoKubo, 2017-12-25:
その後いかがでしょうか。本件は一旦クローズとさせていただきます。また何かありましたら連絡ください。

closed 2017-12-25 by TakakoKubo