MJIrobotics/Tapia-sdk #16: 音声認識のマイクアイコン

IWAMIGINZI, 2017-12-14:
いつもお世話になっております。

音声認識する際、左上に薄緑のマイクアイコンが出現するのですが、
これを非表示もしくは、移動させることは可能でしょうか。

現在、他のボタンとアイコンが被ってしまい、
そのためかボタンの認識の精度が落ちております。

以上、よろしくお願い致します。

TakakoKubo, 2017-12-20:
お世話になっております。
マインアイコンですが、標準装備の動きとなり変更することはできないのですが、
標準SDKを使用せずに、カスタマイズしたメソッドを自身で作成することにより非表示や移動が可能です。以下のTalkActivityでの例です。(ご参考まで)

        sttProvider.setOnStateChangeListener(new STTProvider.OnStateChangeListener() {
            @Override
            public void onStateChange(STTProvider.State newState) {
                switch (newState){
                    case IDLE:
                        if(ttsProvider.getTTSState() == TTSProvider.State.IDLE)
                            tapiaAnimation.setBackground(0);
                        break;
                    case LISTENING:
                        tapiaAnimation.setBackground(R.drawable.gradient_aqua);
                        break;
                    case PROCESSING:
                        tapiaAnimation.setBackground(R.drawable.gradient_defult);
                        break;
                }
            }
        });

IWAMIGINZI, 2017-12-20:
お世話になっております。
標準SDKでは解決できないこと承知致しました。
また、ご回答いただいた件について確認なのですが、
マイクアイコンはtapiaAnimationを使うActivityでしか消せないという認識で合っていますでしょうか。

TakakoKubo, 2017-12-27:
ご返信遅くなり申し訳ありません。。
マイクアイコン画像の非表示については、その認識で合っておりますので、どうぞよろしくお願いいたします。

IWAMIGINZI, 2017-12-27:
お世話になっております。
ご回答いただきありがとうございました。

closed 2017-12-27 by IWAMIGINZI