MJIrobotics/Tapia-sdk #17: Tapia起動時に自作アプリを実行したい

hirokun28, 2017-12-22:
お世話になります。

Tapia-SDKを使用して自作アプリを作成しており、
Tapia起動時に自作アプリが実行されるような設定を行いたいのですが
以下のような現象が起きており上手く行きません。

1. Android Studioからサンプルプロジェクトを参考にした自作アプリをTapia実機で実行
2. Tapiaの設定アプリ→端末→ホームから自作アプリを設定
3. Tapiaを再起動すると自作アプリが立ち上がる
4. Tapiaが人の声か顔（？）に反応すると純正のActivityが割り込み画面が遷移する
5. Tapiaの電源ボタン長押し→ホームボタンを押下
6. 自作アプリが立ち上がり、その後問題なく操作可能

Tapia起動時に必ず一度純正Activityの割り込みが発生してしまいます。
Tapia起動から終了まで常に自作アプリが表示されるような状態が理想です。

何とか実現する方法はございませんでしょうか。
よろしくお願いいたします。

TakakoKubo, 2017-12-25:
現象の確認をさせていただきたいのですが、、
純正のActivityとは、製品版Tapiaで、そのアプリが起動して割り込むという事でしょうか?
そして、毎回起動毎にそのアプリを終了させる必要があるという事ですか?

hirokun28, 2017-12-25:
@TakakoKubo 
説明が分かりづらく申し訳ございません。
おっしゃるとおりでございます。

Tapiaを起動したら自作アプリが常に表示されるよう設定したいのですが、
製品版Tapiaの起動時に下記URLに記載の「待機モードの時に、音に反応した後、
カメラが約8秒間起動し画面に表示」機能が起動してしまい、
ホームに設定していた自作アプリがバックグラウンドに移ってしまう現象を
解決したく質問いたしました。

タピアの状態｜タピア取扱説明書
https://mjirobotics.co.jp/tapia_manual/%E3%82%BF%E3%83%94%E3%82%A2%E3%81%AE%E7%8A%B6%E6%85%8B

TakakoKubo, 2017-12-26:
大変申し訳ありません。
タピアに製品版タピアがインストールされている場合、製品版タピアが起動するように標準で設定されております。よって通常、SDKを使用する場合は、製品版を削除して出荷しております。
何かの手違いにより製品版が入ったまま出荷されてしまったと考えられるため製品版の削除手順をご連絡いたしますので、大変恐れ入りますがそれに従い削除していただくのでよろしいでしょうか?

hirokun28, 2017-12-26:
こちらこそSDKを使う場合は製品版の機能を消してから利用する旨把握しておらずすみません。

製品版の削除手順についてご教示頂けますと幸いです。

TakakoKubo, 2017-12-26:
[unlock.zip](https://github.com/MJIrobotics/Tapia-sdk/files/1586538/unlock.zip)
削除について対応ありがとうございます。
手順は以下の通りです。
①タピアをデバックモードにする
②添付のunlock.zipを解凍する
③windowsｺﾏﾝﾄﾞプロンプト開く
④以下ｺﾏﾝﾄﾞ入力する
　adb shell am broadcast -a malata.adb.root　
⑤adb remount 
⑥ adb shell rm -rf /system/priv-app/tapia　
⑦ adb shell rm -rf /system/priv-app/tapiaUpdate
⑧ adb shell rm -rf /data/app/com.tapia.mji.tapia-1
⑨ adb shell rm -rf /data/app/com.tapia.mji.tapia-2
⑩adb install -r -d unlock.apk　
(先ほど解凍したファイルをたたきます、コマンドプロンプトと同じ場所に解凍した場合は、パスの指定をしないで大丈夫です。)
それではよろしくお願いいたします。また何か質問ありましたら連絡ください。

hirokun28, 2017-12-26:
承知いたしました。
ご対応ありがとうございます。

closed 2017-12-26 by hirokun28