MJIrobotics/Tapia-sdk #18: Maximum Daily Request reached

HiroakiKanda, 2018-01-30:
Androidタブレットでタピアの動作確認をおこなっていると

Maximum Daily Request reached
Please contact MJI to upgrade your plan

と、メッセージが表示されます。
これはなんでしょうか？

どういう条件で出力されますか？
メッセージを出ないようにするにはどうすれば良いですか？

岩谷情報システム 神田

TakakoKubo, 2018-02-01:
タピアの会話に制限があり、規定回数を超えると表示されます。
以下の操作で、利用回数、解除日時を確認できます。
メッセージが出た場合、この利用状況を確認の上すすめていただけないでしょうか。
■確認手順
①SDKサイトにログイン
②左ナビゲーションバーの「マイアカウント」の「アカウントステータス」を選択
・直近STT利用状況/24時間(最大500)
・直近STT利用状況/分(最大20)

HiroakiKanda, 2018-02-02:
お世話になります。

アカウントタイプが有償ライセンス でかつ、
オフラインNLUを使用している状態ですが
それでも制限がかかるのでしょうか？

なぜ？なんのための制限でしょうか？

Please contact MJI to upgrade your plan
とメッセージがありますが
更に上位のプランがあるのでしょうか？

岩谷情報システム　神田

TakakoKubo, 2018-02-02:
いつもお世話になっております。
タピアの音声認識についてですが、音声をテキストに変換する部分はオンラインを使用しています。
その為、一斉にアクセスが集中した場合、音声認識の性能に問題が出てくるため制限をかけております。
また、上位のプランですが、現在のSDKの開発の段階ではそのようなプランはないのですが
製品化の時に必要な場合は相談させていただければと思います。
それではどうぞよろしくお願いいたします。
他に不明点ありましたら連絡くださいますようお願い申し上げます。

MJIrobotics, 2018-04-25:
いつもお世話になっております。本件クローズさせていただきます。また何かありましたら問い合わせ下さいますようお願い申し上げます。

MJIrobotics, 2018-04-25:
MJI

closed 2018-04-25 by MJIrobotics