MJIrobotics/Tapia-sdk #19: TAPIAのI/Oについて

kazkun, 2018-02-14:
TAPIAをとあるサービスに使えないか検討しております。(現在無償登録状態)

まず、
https://tos.mjirobotics.co.jp/sdk/manual/1.1.html にあるSDK利用マニュアル
というのはどちらから参照できるのでしょうか。

また表題でもあるI/Oですが、Bluetooth の表記がありませんが、利用可能でしょうか。
(miniのプレスではBluetoothの記載がありましたので気になりました)

また、Zigbee、NFC、赤外線などはどうでしょうか。将来的な計画でも構いません。

石垣

TakakoKubo, 2018-02-15:
お世話になっております。
昨日、問い合わせいただきありがとうございました。
メールを転送いたしましたが受信できましたでしょうか。

マニュアルですが、大変申し訳ありません。現在、修正をしております。
マニュアルにある情報は、サイト上に掲載しておりますので
大変お手数おかけしますがサイトを閲覧していただけないでしょうか。

なお、BluetoothとNFCは利用出来ます。
Zigbeeと赤外線は現在利用できません。今のところ予定はありません。
それではご検討お願いいたします。

kazkun, 2018-02-15:
ご回答頂きありがとうございます。

メールですが、会員登録の自動送信メールから飛ぶメールは、弊社のメールセキュリティゲートウェイで
偽装していると判定されて届かないようです。またdocomo.ne.jpのキャリアメールも同様です。
gmailは届くもののSPAMと判断され迷惑メールフォルダに直行してしまいます。
また転送頂いたメールは、発行から1時間以上経過していたのか、エラーとなってしまい、
結局何度か試行錯誤して、本来登録したい会社のメールアドレスではないアドレスでの登録となっています。

マニュアルの状況は理解いたしました。引き続きよろしくお願い致します。
BluetoothおよびNFCは可能とのこと、承知しました。

BluetoothはBluetooth LEも利用可能ということでよろしいでしょうか。
またペアリング可能最大数が分かれば教えてください。
NFCはセンサがどのあたりにあるかをお教えいただけますでしょうか。

TAPIAの用途から考えると赤外線はサポート頂くと良いなと思いました。

kazkun, 2018-02-15:
上記に追記で、NFCはType-A/B/F どれに対応するかも回答頂けますでしょうか。

TakakoKubo, 2018-02-19:
お世話になっております。
Bluetoothですが、仕様上は最大7台となっておりますが
データの処理が多いデバイスを使用した場合は、必ずしも最大数を接続できるとは限らないと思われます。
また、NFCのセンサはタピアの頭上についております。Type-A/B/F をサポートしています。
よろしくお願いいたします。

kazkun, 2018-02-19:
ご回答ありがとうございました。

BLE(BluetoothLE)対応は如何でしょうか。
接続しようとしている機器としては、今のところIoT的なものでトラフィックはさほど多くないです。

TakakoKubo, 2018-02-20:
お世話になっております。
Bluetoothは、Version	:4.0　です。
対応していると思いますのでどうぞよろしくお願いいたします。

kazkun, 2018-02-22:
お返事遅くなりました。
承知いたしました。ありがとうございます。

MJIrobotics, 2018-04-10:
クローズいたします。

closed 2018-04-10 by MJIrobotics