MJIrobotics/Tapia-sdk #20: Android5.1 Storage Access Frameworkについて

tkyk0520ossc, 2018-04-09:
現在Tapiaアプリで、Tapiaが撮影した画像、動画をTapiaに挿入されたmicroSDカードへ保存する機能を開発しているのですが、普通のAndroid端末でテストした際はStorageAccessFrameworkが正常に動作し、SDカード内の画像削除や動画の作成などが行えたのですが、Tapia実機でテストすると失敗してしまいます。Tapiaには、StorageAccessFrameworkは入っていないのでしょうか。
また、TapiaでのmicroSDカードへのデータ保存方法をご教授頂けると幸いです。
宜しくお願い致します。

MJIrobotics, 2018-04-10:
現在状況の確認中です。確認後ご連絡いたしますので少しお時間いただきたくお願いいたします。どうぞよろしくお願いいたします。

tkyk0520ossc, 2018-04-10:
早急なご対応ありがとうございます。
以上の質問に加えて、
独自に開発したTapiaアプリからmicroSDカード内のデータ削除方法もご教授頂けると幸いです。
何卒宜しくお願い致します。

MJIrobotics, 2018-04-24:
いつもお世話になっております。
MJIです。
検証事項等が多々ありましたがゆえこの度はお時間いただきました事大変申し訳ありませんでした。
さてご質問いただきました外部microSDカードへの保存ですが、
原則としてアプリからの保存、削除は出来ないようです。出来るのは読込のみ可能となります。
大変恐れ入りますが、保存についてはタピア本体またはインターネット接続上でネットワークドライブの活用等を検討していただけたら幸いに思います。
それではどうぞよろしくお願い申し上げます。また何かまた不明点質問等ありましたらご連絡くださいますようお願いします。

MJIrobotics, 2018-04-25:
いつもお世話になっております。本件クローズさせていただきたいと思います。また何かありましたら問い合わせ下さいますようお願い申し上げます。

closed 2018-04-25 by MJIrobotics