MJIrobotics/Tapia-sdk #21: 中国語の発話について

jts-kitanaka, 2018-05-01:
お世話になります。

TAPIAに中国語で発話をさせたいです。
例えば「こんにちは。私はtapiaです。」と中国語で発話させてみようとしましたが

ttsProvider.sayIn("你好,我是tapia", Language.LanguageID.CHINESE_TW);
ttsProvider.sayIn("你好,我是tapia", Language.LanguageID.CHINESE_CN);
ttsProvider.sayIn("你好,我是tapia", Language.LanguageID.CHINESE_HK);

上記３つのコードのうち
　LanguageIDにCHINESE_TWを設定したもののみ発話されました。
　
tapia-sampleのLanguage.classにはLanguageIDとして
下記３行の中国語らしい値がありましたが
　CHINESE_TW("臺灣華語"),
　CHINESE_HK("廣東話"),
　CHINESE_CN("官話"),
　
タピアで発話できるのは、CHINESE_TW 台湾国語（繁体字？）のテキストのみなのでしょうか？

MJIrobotics, 2018-05-02:
いつもお世話になっております。
タピアは中国語ではLanguageID.CHINESE_TWのみに対応していますので
中国での発話はこちらの使用をお願いいたします。
どうぞよろしくお願いいたします。

MJIrobotics, 2018-05-31:
クローズします。

closed 2018-05-31 by MJIrobotics