MJIrobotics/Tapia-sdk #23: Permission Error (無償アカウントで実行)

tkyk0520ossc, 2018-06-27:
無償アカウントで、アカウントを登録したのですが、
TOSチュートリアル通りに設定を行い、いくら実行しても、
以下のようなエラーがTapia画面に出力されてしまいます。
一度発声するのですが、その直後にエラーダイアログが表示される状態です。
______________________________________________________________
Permission Error
Your permission is invalid or expire,please contact mji for more infomation
______________________________________________________________

確実に動くはずのテスト用アカウントを一時的にお借りする事は可能でしょうか。
宜しくお願い致します。

MJIrobotics, 2018-06-28:
お世話になっております。
この度はご不便をおかけして申し訳ありません。昨日サーバー移行により一部エラーが発生したいたことが原因かもしれませんが、現在も発生する場合はアカウントの確認したいので登録したメールアドレスを教えていただけないでしょうか。どうぞよろしくお願いいたします。

MJIrobotics, 2018-07-02:
お世話になっております。
その後「Permission Error」は発生していますでしょうか。もしもまだ表示される場合ありましたら連絡下さいますようお願いいたします。

closed 2018-07-02 by MJIrobotics

tkyk0520ossc, 2018-07-03:
返答が遅れて申し訳ありません。
確かめてみます。
ご返答ありがとうございます。

登録したメールアドレスは
tkyk0520@gmail.com
でございます。
宜しくお願い致します。

＿＿＿＿＿＿＿＿＿＿＿＿＿＿＿＿＿＿＿＿
株式会社アウトソーシングテクノロジー
企画開発課

安齋　聖之
anzai takayuki

Tel：08044413560
mail：t-anzai@ostechnology.co.jp

＿＿＿＿＿＿＿＿＿＿＿＿＿＿＿＿＿＿＿＿＿