MJIrobotics/Tapia-sdk #25: ビルド後に処理が落ちてしまう

yukiito-realsys, 2020-02-06:
Android5.1のタブレット上にて実行を行うと
ビルド完了後にエラーで落ちてしまいます。

何か設定がおかしいのでしょうか？

下記エラー文をお送りいたします。

`
E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.tapia.mji.demo, PID: 15044
    java.lang.RuntimeException: Unable to create application com.tapia.mji.tapialib.TapiaApp: java.lang.NumberFormatException: Invalid long: "ystemUI_V100R001C00B003"
        at android.app.ActivityThread.handleBindApplication(ActivityThread.java:4859)
        at android.app.ActivityThread.access$1900(ActivityThread.java:165)
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1442)
        at android.os.Handler.dispatchMessage(Handler.java:102)
        at android.os.Looper.loop(Looper.java:135)
        at android.app.ActivityThread.main(ActivityThread.java:5593)
        at java.lang.reflect.Method.invoke(Native Method)
        at java.lang.reflect.Method.invoke(Method.java:372)
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:960)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:755)
     Caused by: java.lang.NumberFormatException: Invalid long: "ystemUI_V100R001C00B003"
        at java.lang.Long.invalidLong(Long.java:124)
        at java.lang.Long.parse(Long.java:363)
        at java.lang.Long.parseLong(Long.java:353)
        at java.lang.Long.parseLong(Long.java:321)
        at com.tapia.mji.tapialib.TapiaApp.hasNewUi(TapiaApp.java:536)
        at com.tapia.mji.tapialib.TapiaApp.onCreate(TapiaApp.java:249)
        at android.app.Instrumentation.callApplicationOnCreate(Instrumentation.java:1012)
        at android.app.ActivityThread.handleBindApplication(ActivityThread.java:4850)
        at android.app.ActivityThread.access$1900(ActivityThread.java:165) 
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1442) 
        at android.os.Handler.dispatchMessage(Handler.java:102) 
        at android.os.Looper.loop(Looper.java:135) 
        at android.app.ActivityThread.main(ActivityThread.java:5593) 
        at java.lang.reflect.Method.invoke(Native Method) 
        at java.lang.reflect.Method.invoke(Method.java:372) 
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:960) 
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:755) `

TakakoKubo, 2020-05-18:
ご返信遅くなり大変申し訳ありませんでした。。
タブレットで使用する場合、タピアが回転をするメソッドがありましてそれをオフする必要があります。その設定はしているかご確認をお願いいたします。
スタートアクティビティのTapiaAppのEnabelをfalseにします。

TakakoKubo, 2020-06-11:
その後、ビルド完了後のエラーについて、状況いかがでしょうか。
一旦クローズいたしますので、解決しない場合ございましたら是非ご連絡くださいますようお願いいします。

closed 2020-06-11 by TakakoKubo